HaddingtonDynamics/Dexter #11: Return FPGA data

JamesNewton, 2018-07-28:
Get XYZ, cosine, other data back from the FPGA through DexRun.c to DDE.

While there may be some room in the return status data, many possible return data items may be of use in the future. Use [read_from_robot](../wiki/read-from-robot) for #XYZ "file"? Or use #FPGA and an address number for general purpose access?

JamesNewton, 2018-09-13:
#XYZ was added here: https://github.com/HaddingtonDynamics/Dexter/commit/0a55652894b1e445b9233f22fe0c8e782c6c62a2 but the actual code is here: https://github.com/HaddingtonDynamics/Dexter/commit/e6db50da946176123e191e9af6660a318f240489 starting at line 2118

> Requires  gateware release of 8/26/2018 (no version number yet)
> 
> -New register constants for measured angles
> -New structure called pos_ori_mat (short for "position orientation matrix"):
>     -access each element in the form of A.r0.c0, r is the rows 0-3 and c is the columns 0-3
> -New function J_angles_to_pos_ori_mat():
>     -converts 5 joint angles to a coordinate system of the last joint in the form of a pos_ori_mat
>     -origin is located at point 4 i.e. centered on J5's axis, above the differential at height L5 from the axis of J4.
>     -the orientation aligns with the robot's coordinate system when in home position: Z is aligned with the J5 axis positive upward, Y is 
>       in the direction of the end-effector, and X the cross product of of the Y and Z vectors (in that order, right hand rule)
> -pos_ori_mat_to_string(): takes inputs of a pos_ri_mat and a char array, fills the char array with the matrix string
> -psuedo-file functionality added to read_from_robot
>     -file paths that start with '#' will return file contents for particular data without creating an actual file
>     -#XYZ returns the current measured pos_ori_mat
>     -#measured_angles returns the current measured angles
>     -more functionality on the way

Note that measured angles are also being returned in the [standard status oplet 'g'](../wiki/status-data) via this commit on the TDint branch: https://github.com/HaddingtonDynamics/Dexter/commit/1ca9251b47468d9841713ec89b62e91050125188

closed 2018-09-13 by JamesNewton

cfry, 2018-09-14:
Requires gateware release of 8/26/2018 (no version number yet)

-New register constants for measured angles
-New structure called pos_ori_mat (short for "position orientation matrix"):

Check with James W but I think this might be a "pose".

Let's be careful here with terminology, inventing names,

and consistency.

See Doc pane/Articles/Dexter Kinematics/Glossary

This needs work.

-access each element in the form of A.r0.c0, r is the rows 0-3 and c is the
columns 0-3
-New function J_angles_to_pos_ori_mat():
-converts 5 joint angles to a coordinate system of the last joint in the
form of a pos_ori_mat
-origin is located at point 4 i.e. centered on J5's axis, above the
differential at height L5 from the axis of J4.
-the orientation aligns with the robot's coordinate system when in home
position: Z is aligned with the J5 axis positive upward, Y is
in the direction of the end-effector, and X the cross product of of the Y
and Z vectors (in that order, right hand rule)
-pos_ori_mat_to_string(): takes inputs of a pos_ri_mat and a char array,
fills the char array with the matrix string

by pos_ri_mat did you really mean pos_ori_mat ?

Still need terminology standardization.

-psuedo-file functionality added to read_from_robot
-file paths that start with '#' will return file contents for particular
data without creating an actual file

Did you really mean to say "will return a string even though there's

no actual file associated with the "file name" ?

-#XYZ returns the current measured pos_ori_mat
-#measured_angles returns the current measured angles
-more functionality on the way

It looks like you're using # as a prefix to mean

"this is just pure date, there's no actual file associated with it.

But unix does't need that # ie   /dev/null

so wouldn't it be best to follow the Unix convention here?

We might have some data that we decide to implement as

a standard file, then realize later it needs to be "dynamic".

If we DON'T have the "#" prefix, then we don't have to

change user code referencing it, we're just "swapping" the

underlying representation. What does the user care if

its a file or dynamically created?

>

JamesNewton, 2018-09-14:
_prior reponse edited to clarify what was quoted and what was a new comment. email replies don't support markdown (which makes sense, as email is rarely written in markdown) so clarity is better if comments are made on the issue instead of via email._

Since the [read from robot](../wiki/read-from-robot) oplet was written and documented as returning file data, it's later use to return dynamically generated strings was thought of as dynamically generated files. It is, as noted, just returning a string.

The # symbol was chosen specifically because the common file systems (Linux, Mac, ,Windows) do not allow it, and therefore, there is no chance of blocking access to a real file. If we had chosen e.g. _ as our flag that this was a request for a dynamically generated "file" then it would be impossible to access a file named e.g. _test. This ensures that the user will always have access to every file on the file system irrespective of it's name, or what dynamically generated files are supported in the firmware and this new usage of the mechanism to return strings can't possible interfere with it's original function.

Note: The read from robot opet was chosen for this because the string could easily be longer than one socket packet. Strings were desirable because they avoid having to interpret binary data which is a problem for some languages and programmers, myself included.

cfry, 2018-09-14:
> Since the read from robot <http://../wiki/read-from-robot> oplet was

Ahh, I see. We can imagine /foo  as a file and /#foo as
extra data. In the unix case, maybe they control carefully
the "dynamic" names so as not to insure conflicts,
ie you can't create a file named /dev/null,
unix would just error if you tried.
But we can't so easily do that.