HaddingtonDynamics/Dexter #5: Startup repeatability

JamesNewton, 2018-07-28:
Having a way for Dexter to find "home" position automatically would be a wonderful help. This will both allow it to accurately interact with a work surface or other object via "dead reckoning" (no need to touch or be guided to a starting point). It will also removed the need to calibrate Dexter on startup because the prior calibration can be applied immediately. 

Currently, the solution to this is to add a "dial" or little knob to each motor shaft. The robot is set to a perfect zero position once during manufacture (with a test stand) then the dials are installed and their correct positions marked. While powered off, the robot is move close to the home position, then each dial is turned until it lines up with the known home position.

Automated methods of solving this problem include:
- An index pulse on each joint encoder. The issue with this is the cost of an addition sensor and extra size of the disk. It may be possible to get the same effect by filling in one of the regular slots with a semi-transparent material (it looks like "Elmers Washable Clear Glue" works well) so that the slot is still there, still the same width, but doesn't "open" as far as the other slots. On power up, each joint would be "wiggled" until the index slot is detected. Redundancy is provided by the fact that each sensor should see the same effect, some fixed number of degrees apart.
- A mechanical home switch on each motor shaft. Just like the dials, the arm would be preset to near the home position, then on power up, each joint would be "wiggled" until the home switch closes.

JamesNewton, 2018-09-04:
This works on my Dexter, needs to be tested on other Dexters and then possibly integrated into the DexRun Firmware:

````

/*
Find index pulse by putting a little Elmers washable clear glue #E305 in one of the slots.
The glue causes the light to be reduced and therefor a "smushed" eye during cal.
Because there are two sensors, sin and cos, which are 90' out of phase, we should see
one eye smushed vertically, and then another smushed horizontally. The sensors are NOT
right next to each other, so the smushed eyes are some number of slots apart from one 
another. 
*/

var joint = 1 - 1
var joints = 5 //const number of joints
//var joints_slots = [200, 135, 157, 115, 100] //number of slots in each disk
//joint sensor offset angles: 
//Joint1 Index X:3.7', Y:-5', delta:-8.7. Slots X:50, Y:50
//Joint2 Index X:-3.7', Y:-13.8', delta:-10.1. Slots X:34, Y:34
//Joint3 Index X:25.4', Y:-9.6', delta:-35. Slots X:39, Y:39
//Joint4 Index X:-7.8', Y:30.5', delta:38.3. Slots X:29, Y:29
//Joint5 Joint5Index X:2.3', Y:-12.8', delta:-15.1. Slots X:25, Y:24 (actual angles are WAY off because index glue was misplaced.
//var slots = joints_slots[joint]
var max_speed = 25 // degrees per second 
var max_accel = 0.0000162 //degrees per second per second? 0.000129 jerks
var min_threshold = 0.22
var avg_over = 1.2
var angle = -35
var angle_step = 0.2
var ramp_angle = 5 //should be related to velocity and accelleration

var angles = [angle / 2, -angle / 1.5, angle / 1.2, -angle, angle]
var step_angles = [angle_step / 2, -angle_step / 1.5, angle_step / 1.2, -angle_step, angle_step]
var start_angles = [0, 0, 0, 0, 0]
var sin = []
var cos = []
for (var i = 0; i < joints; i++) {
	sin[i] = eval("Dexter.J" + ( i + 1 ) + "_A2D_SIN")
	cos[i] = eval("Dexter.J" + ( i + 1 ) + "_A2D_COS")
	start_angles[i] = angles[i] + (angles[i] > 0 ? ramp_angle : -ramp_angle)
    //if ((i % 2)) { start_angles[i] = -start_angles[i]} //opposite direction on every other joint
    }

var step = 0
var steps = (Math.abs(angle) / angle_step) * 2 //negative and positive
//var steps_per_degree = 1 / angle_step
//var degree_per_slot = 360 / slots
//var steps_per_slot = steps_per_degree * degree_per_slot
var x = []
var y = []
var xs = [[], [], [], [], []]
var ys = [[], [], [], [], []]
var as = [[], [], [], [], []]
var eye = []
var index = []
var max_adc = Math.pow(2, 12)
    min_threshold = min_threshold * max_adc

var x_max = [max_adc, max_adc, max_adc, max_adc, max_adc]
var x_avg = [max_adc / 2, max_adc / 2, max_adc / 2, max_adc / 2, max_adc / 2]
var x_maxs = [[], [], [], [], []]
var x_max_min = [[0, max_adc], [0, max_adc], [0, max_adc], [0, max_adc], [0, max_adc]]
var x_maxed = [false, false, false, false, false]
var y_max = [max_adc, max_adc, max_adc, max_adc, max_adc]
var y_avg = [max_adc / 2, max_adc / 2, max_adc / 2, max_adc / 2, max_adc / 2]
var y_maxs = [[], [], [], [], []]
var y_max_min = [[0, max_adc], [0, max_adc], [0, max_adc], [0, max_adc], [0, max_adc]]
var y_maxed = [false, false, false, false, false]

var xsp //so we can mess with it after the job runs. 
[... 178 lines cut ...]
            new_points[i][1] = -(info.scale[1]*points[i][1]+info.center[1])+info.dim[1]
            }
        return new_points
        }
    
    //returned subfunction draws polylines between given points
    function draw(points, size=1, color="black", border="white", layer="layer1") { 
      append_in_ui(title+"_id",
		svg_polyline({html_class: layer
        , points: trans(this, points), color: color, width: size})
		)
      }
    //the layer parameter in the returned subfunction allows plotted or drawn items to be grouped, 
    //then accessed. e.g. erased for animations
    function remove_layer(layer) {
    	$("."+layer).remove()
        }
        
    let result = {scale: scale, dim: dim, center: center, plot: plot, draw: draw, remove_layer: remove_layer}
    return result
}

//see also:
//https://github.com/HaddingtonDynamics/Dexter/blob/TDint/Firmware/DexRun.c#L2077
````

JamesNewton, 2019-05-19:
In order to find position faster after power up in an unknown position, a set of index pulses, instead of just one, could be used.

If the gap between index pulses is different on the two sides of the center "home" index pulse, then the correct direction to rotate to reach home. E.g. if there are an odd number of regular sized eyes between index pulses on one side of home and an even number on the other side, a small movement in any direction from any position should tell us which way to move to reach home.

If the count of regular sized eyes between index pulses increases as you move away from home, then the exact position can be found faster. E.g. on one side, there might be 2 then 4, 6, 8 regular eyes between indexes and on the other side, 3, 5, 7, and 9.

cfry, 2019-05-20:
Clever!
_____
Lots of games could be played here.
For instance, the idea of partially covering a slot means
we effectively get 2 bits from it, ie "that it is a slot"
and that it is "partial or full".
But can we get more bits out of it? ie
1/4 covered, 1/2 covered, 3/4 covered?
and apply that to James N's trick to tell
how far a slot is from home?
Maybe this interferes too much with the normal accuracy.
A 2nd row of slots, closer to disc center, lower rez but
more bits per slot as I describe above?
Transparently colored slots?
Just throwing some more cards on the table.

JamesWigglesworth, 2019-07-19:
Progress has been made on this issue.

We put what we're calling "Index Eyes" on each of the 5 code disks.
Index Eyes are slightly more narrow slits on the already existing slits of the code disk.
They can be detected against a set of normal eyes.
There is a unique pattern to the spacing between all of the index eyes such that once the robot finds just two of them it can use the spacing to know exactly where it is.
This information can then be used to move to the 'true home' position and re-define where 0 degrees is.

Here is a video of the first successful prototype of this system:
https://photos.app.goo.gl/KrK1dw75fzCAtEYJ8

There will still be some time before this get integrated into the production version of Dexter.

Here is the code that was used in the video:

````
/*
Written by: James Wigglesworth 
Started: 5_21_19
Updated: 7_17_19

Overview:
#A User Input
#B Globals
#C init_globals()
#D init_robot()
#E find_idx_eyes()
	#E1 is_idx_eye()
    #E2 AdcCenters string conversions
#F set_found_position()
#G Find_Idx_Eyes Job Def
#H Test_All_States Job Def
*/

/*
w, 42, 832
make_ins('w', 42, 832)
set eye number
set commanded angle
*/

//#A User Input:
var initial_direction = [-1, -1, -1, 1, -1]
var slow_step_size = [1000*_arcsec, 1000*_arcsec, 1000*_arcsec, 1000*_arcsec,  1000*_arcsec]//[0, 0, 0, 0, 100*_arcsec]
var rapid_steps_per_eye = [15, 15, 15, 15, 15]
var idid = 0
var n_eyes = [200, 180, 120, 113, 100]
var max_eye_spacing = [30, 30, 30, 30, 14]

var idx_eye_to_cal_offset = [5, -6, 8, -5, 0]
// out(Vector.subtract([255, 255, 255, 255, 255], true_home_eye))
var post_cal_info = JSON.parse(file_content("C:/Users/james/Documents/dde_apps/2019/_Code/Misc/EncoderIdeas/on_bootup_info"))
var tweek_angles = [5.5125, 0.03333333333333333, 0.0225, 0.013125, 0.046875]

/*
var temp = [
	Dexter.dexter0.robot_status[Dexter.J1_ANGLE],
    Dexter.dexter0.robot_status[Dexter.J2_ANGLE],
    Dexter.dexter0.robot_status[Dexter.J3_ANGLE],
    Dexter.dexter0.robot_status[Dexter.J4_ANGLE],
    Dexter.dexter0.robot_status[Dexter.J5_ANGLE]
]
*/
//#B Globals:
var deg_per_eye
var step_size
var start_time
var start_EyeNumber
var old_EyeNumber
var old_RawEncoder
[... 1840 lines cut ...]
        	rs = this.robot.robot_status
            PID_deltas = [
            	rs[Dexter.J1_PID_DELTA],
                rs[Dexter.J2_PID_DELTA],
                rs[Dexter.J3_PID_DELTA],
                rs[Dexter.J4_PID_DELTA],
                rs[Dexter.J5_PID_DELTA]
            ]
            
            out(PID_deltas)
        }
    ]
})

new Job({
	name: "PID_move_by_Deltas",
    inter_do_item_dur: 0.01,
    show_instructions: false,
    do_list: [
    	function(){if(!PID_deltas){dde_error("PID_deltas is undefined. Run PID_Deltas Job.")}},
    	Dexter.pid_move_all_joints(Vector.multiply(1, PID_deltas))
    ]
})

````

JamesNewton, 2019-11-19:
With this Find_Index_Pulses system, this issue is effectively closed for the HDI robots.

closed 2019-11-19 by JamesNewton