HaddingtonDynamics/Dexter #6: Cycloidal Wave transmission

JamesNewton, 2018-07-28:
The Harmonic Drives used in current Dexter versions are difficult to source, expensive and may have issues with backlash due to side loads. Cycloidal Drives hold great promise: It may even be possible to 3D print them with a high end printer like the [Markforge](https://markforged.com/)

If you have experience with this, we want to hear from you.

https://en.wikipedia.org/wiki/Cycloidal_drive

JamesNewton, 2018-07-28:
We are starting to have some success with this design. 

https://twitter.com/HDRobotic/status/1017916434230304769

A new version with better structure to prevent warping under side loads is on the way.

BryanHaven, 2018-10-02:
Would printing this on a Form2 in one of the engineered resins be useful?

JamesNewton, 2018-10-02:
Thanks Bryan! The cycloidal drive design we are working on actually has bearings and other parts inserted during the print job, which really wouldn't work with the liquid resins. They also probably need the strength of the Onyx carbon fiber filament from our Markforged printers. 

There are other parts that it might be interesting to see on a resin printer, such as the code disks. If you want to try one of those, check the STL files at:
https://www.thingiverse.com/thing:2108244/files
and search the page for "CodeDisk". The DIFF ones probably aren't a good test, as they are pretty low count, but the base, end arm, or pivot disks would be interesting to try. The key is that the resolution must be very high, and the material totally opaque.

BryanHaven, 2018-10-02:
Hi James,

I’ve been reading up on the cycloidal drives and was curious if you’re using a dual plate or is vibration not an expected issue? Any concerns regarding lateral forces on the output?

Thanks for the info,I’m looping in Stephen who has the Form2. 

I think the Grey Pro resin would do the trick for the encoders. Just a matter of getting the supports right. 

Stephen, what are your thoughts?

Bryan

JamesNewton, 2018-10-02:
I've started a new issue #39  to continue with the idea of using a resin printer for the encoder disks.

@JamesWigglesworth is the guy to ask about the design of the cycloidal drive, but as I understand it, there are 3 plates, back, middle and front, because there are actually 2 drives in one unit: A 5 and a... 3? More information here:
https://hackaday.com/2018/08/24/a-peek-at-the-mesmerizing-action-of-a-cycloidal-drive/#comment-4976153

(fyi, when you reply via email, it does post that publically here. I mention because some people aren't aware of that).

BryanHaven, 2018-10-02:
I am Stephen and my contribution to the conversation is I have a form2 and
have used a lot of resin.

I attached a Form2 file (readable with preform, downloadable
https://formlabs.com/tools/preform/) of how I'd to the print of the Diff
and Pivot disks.  Light sanding on the back and outside and away you go.
Nice and simple.  I hope.  I'm not an expert on encoders, but I understand
that the spaces need to be really, REALLY uniform, so I'd want some way to
verify that.

I have no idea how to do that.

What I can do is send you a gray normal print of it in the next week or so
that you can evaluate.

The printing cost is going to be minimal (about $1.80) and shipping to a
US address about $7.20 (flat rate).  Unless/until this becomes a bigger
thing with more prints I'm happy to do this for the good fo the community
and our future robot overlords.

JamesNewton, 2018-10-02:
Thanks Stephen / Brian. It would be great if you could respond with that on:
https://github.com/HaddingtonDynamics/Dexter/issues/39
 instead.