HaddingtonDynamics/Dexter #33: Change mode with SetMode oplet

JamesNewton, 2018-08-07:
To make it very very easy to change modes (keep, follow, protect) add a new [SetMode](https://github.com/HaddingtonDynamics/Dexter/wiki/set-parameter-oplet) sub-item to set all the FPGA parameters in one shot.

cfry, 2018-08-09:
See DDE Ref Man/Robot/Dexter/Dexter Instructions/
set_follow_me,
set_force_protect
set_keep_position
set_open_loop

JamesWigglesworth, 2018-08-09:
This is talking about those exact functions but in the form of single oplet instructions to make it easy on the non-dde user.

JamesNewton, 2018-08-23:
@kgallspark @JamesWigglesworth 
As we discussed the other day, it's silly to have specific set modes. It's better to simplify the setting of the values in common ways. These can still be adjusted with individual 'w' or 'S' commands after the main command, but as long as our new 'S' command can reproduce the existing DDE mode sets when given the correct parameters, it will satisfy the letter of the Monty request and also give users a tool that they can use to set "submodes" that smoothly transition from one preset mode to another and hopefully will give them all sorts of different capacities for load and "feel" when moving the arm.

Take a look at this spreasheet which I made from DDE's mode set commands. 
https://docs.google.com/spreadsheets/d/1bf2u-hSuzWSXB12lu0_LHWHDlJ-iFqffg0sEph-rPU4/edit#gid=0

It looks to me like there are 11 parameters that cover all the variations between the modes.  I've highlighted those with the tan colored lines. So, for example, if you pass in a DIFF_FORCE_SPEED_FACTOR_ANGLE of 8, that's what DDE would do to put you in "Follow" mode, if you pass in a 3, that's like "Protect" mode. But now, you can choose other values. I can see that being more useful with the PID, Friction, and Force Decay values. 

We can then document for them some general guidelines. Like "If the arm starts to shake with a heavier load, increase PID XYZ." Or whatever.

Given the size of the socket buffer, that /might/ just fit, but it would be really good to try to combine / reduce the numbers. For example, do we really need 4 different Friction settings, or can we set Base, XYZ, and Wrist like we do for PID? Or do we really need separate Base and XYZ PID values? Base and XYZ seem to always be set the same.

Questions:
1. Are there other parameters we should include?

2. Are there settings we should not be changing? e.g. is DIFF_FORCE_MAX_SPEED needed? (I already pulled all the timebase stuff)

3. Are there settings we can reliably calculate from other settings or via a new parameter? E.g. can "Load Mass" be used to calculate the PID settings?

4. What are the default settings for the missing items in the columns to the right? I've marked those with red question marks.

JamesNewton, 2018-08-25:
Progress:
https://github.com/HaddingtonDynamics/Dexter/commit/f0d9fa772ba6c3eee979e62a071bca487a084c21

cfry, 2018-08-26:
> f0d9fa7

My next release will have support for Dexters of an unlimited joint count
in some cases, and at least 7 in others.
I know some programming by example uses of Dexter have used the
width of the grippers manually manipulated (j7) and probably J6 as well.
In one case at the above URL, there is a link length section
that takes just 5 numbers and a
"fric" setting that takes just 5 numbers.
I'm nervous that we'll paint ourselves into a corner and make
backward compatibility difficult in the face of expansion.
In general, this problem isn't really solvable. We will often
have to put some stakes in the ground and live with them.
But at least for high level cmds like "move_all_joints",
its now not limited in the number of joints it accepts
(err, except packet size I guess).

JamesNewton, 2018-09-04:
Release:
https://github.com/HaddingtonDynamics/Dexter/commit/f0d9fa772ba6c3eee979e62a071bca487a084c21
effectively provides this ability. From the release notes:

Line 4343: Add SetParm ("S") sub-command "Ctrl" which takes named parameters and quickly sets commonly used values. The named parameters are:
- `PIDP # # # ` - sets the PID P value for the BASE to the first number, the END, and PIVOT to the next, and ANGLE and ROT to the last. This controls how hard the arm reacts to an error in the joint position. Too high a value may cause it to shake, too low will cause inaccurate positions.
- `Frict` # # # # # ` - sets each joint friction values. These are floating point. This controls how hard it is to move any joint.
- `Diff # ` - sets the DIFF_FORCE_SPEED_FACTOR_ANGLE and DIFF_FORCE_SPEED_FACTOR_ROT to the specified value
- `FMul # ` - sets SPEED_FACTORA. Multiplier (coefficient / master gain) for the force calculator. 
- `Decay # ` - sets the *_FORCE_DECAY for all joints to the same number. This is how hard the arm tries to return to the commanded position.
- `Cmd # ` - sets the command register. 

For example: An S command with 
"Ctrl Diff 3 FMul 10 PIDP 1045351628 1045351628 1022739087 Frict 2 3 9 15 15 Decay 9000 Cmd 12448" 
would set a Protect mode and 
"Ctrl Diff 8 FMul 30 PIDP 0 0 0 Frict 5 5 5 15 15 Decay 0  Cmd 12448" would set a "Follow" mode. 

Note that the total command length, including the job and instruction number, start time, end time, command letter, and all the parameters, can not exceed 128 character. That leaves approx 90 characters for the Ctrl command string. 

Future named parameters may simplify settings. E.g. a `Mass` value could possibly set the PID drive based on the mass of the object being lifted.

cfry, 2018-09-04:
In the example, it would be helpful if it
included a complete example from DDE, ie
something that starts with
make_inst("S" ....)

JamesNewton, 2018-09-12:
https://github.com/HaddingtonDynamics/Dexter/commit/361d65dfb0b22d1907449a4e814349a7571e0577
Just makes this a bit easier by ignoring JSON formatting characters. E.g. instead of having to do 
`PID_P 1,2,3`
you can do 
`[PID_P: 1, 2, 3]`
and the firmware won't see any difference.

JamesNewton, 2018-09-12:
> In the example, it would be helpful if it included a complete example from DDE, ie something that starts with make_inst("S" ....)

This normally wouldn't be done from DDE as DDE has other ways of setting it, but, one example (from the spreadsheet above) would be:

`make_inst("S Ctrl Diff 8 FMul 30 PIDP 0 0 0 Frict 5 5 5 15 15 Decay 0  Cmd 12448")`

JamesNewton, 2018-10-08:
Our documentation didn't make it clear that the numbers for the PIDP values are actually IEEE 32 bit floating point numbers, being passed in from DDE as integers. To make reasonable changes to these values, we need to develop, document, and then automate the process of starting with the desired floating point value, converting that to a binary representation ala IEEE 754 and then convert that to a decimal number to send to DexRun. Once we have that process, we can amend the Ctrl command to accept Floating Point values directly.

https://en.wikipedia.org/wiki/Single-precision_floating-point_format#IEEE_754_single-precision_binary_floating-point_format:_binary32

JamesNewton, 2018-10-13:
Commit https://github.com/HaddingtonDynamics/Dexter/commit/e3cccb88cf9b1671c5b2f13c77d1ee00a967a528 Adds support for seting the PID P values using floating point, via the [J#_PID_P parameters to the SetParam oplet](https://github.com/HaddingtonDynamics/Dexter/wiki/set-parameter-oplet) e.g. `S J1_PID_P 0.2` To avoid breaking backward compatibility, the floating point support was NOT added to the Ctrl parameter or the 'w' oplet / write command, but instead:

To support single command mode setting (and much more) the [RunFile parameter for the SetParam oplet](https://github.com/HaddingtonDynamics/Dexter/wiki/set-parameter-oplet) is a replacement for Ctrl with several advantages:
1. It has no length limit.
2. It can perform ANY full S or w command without abbreviation.
3. It can perform ANY standard oplet/command period... even moves. Even calibrate. Or demos. Or running other files. 
4. It can still do all that in one command because it pulls the instructions from a file on the robot e.g. `FollowMode.make_ins`

For now, both RunFile and Ctrl are still available.

closed 2018-10-13 by JamesNewton