HaddingtonDynamics/Dexter #35: Startup / Home Dexter in Positions other than Straight up

JamesNewton, 2018-09-04:
In some applications, it is not possible to have Dexter setup in a completely straight or vertical position. It would be good to be able to start Dexter in an offset position.

cfry, 2018-09-04:
I am expecting most jobs to assume Dexter could be
in any position at startup and thus they
should move dexter to a "starting position" before
doing the domain-specific activity of the job.
Do to tool paths and things that "might be in the way",
this might mean moving dexter twice,
ie once "straight up" as far as it can go"
then, now that its theoretically clear of the workspace,
over to a high "x, y start position".
Then from THERE we start the job's specific movements
in earnest.

One trick to normally avoid obstacles is
before moving TO somewhere on the work surface,
first move up to a neutral high position.
This won't always work, but is a good heuristic.

JamesNewton, 2018-10-08:
b5c8c9815294405c8ec51c3ce6bdb3ed3433e139 is an effective workaround for this but it will not work if you are using DDE to do kinematics. (e.g. Dexter.move_to )

commit b5c8c981 (2018-09-27): Add StartPosition.txt offset to joint angles

This should allow kinematics to work when the robot is started at a position other than totally upright. E.g. with a file in the share folder named StartPosition.txt containing:
[ 0, 0, 324000, 0, 0 ]
the joint 3 start position can be 90 forward and all XYZ moves via the "M" oplet, and the #POM data pseudo file via read_from_robot will still work. 

Note that the values are in arcseconds, i.e. degrees * 3600. In our example, 90 * 3600 = 324000.

This should close #35
closed 2018-11-06 by JamesNewton