HaddingtonDynamics/Dexter #37: Can't directly CAT5 between PC and Dexter since DHCP enabled

JamesNewton, 2018-09-14:
When Dexters had a fixed ip address, you could just connect a CAT5 cable between the PC and Dexter and it would work. There were problems if you enabled your WiFi and it went up on a different subnet, but that was fixable by changing Dexters IP.

When Dexter's firmware was changed to use DHCP, it was MUCH better for use on a real network, but that requires that we have a router with DHCP for Dexter and the PC to connect into. Carrying the extra router is a pain. It would be nice to have Dexter start in DHCP, and then if that isn't found, fall back to a static IP.

JamesNewton, 2018-09-14:
A partial solution is to set a fixed IP which is always up, in addition to the DHCP address. After SSHing into  Dexter:

`nano /etc/dhcp/dhclient.conf`

and comment IN the alias section, then edit it to pick a static address that won't cause issues on your network. e.g. the fixed ip for the robot. I used one on my 192.168.0 network, but you should be able to use any local network address that isn't the same as your PC or anything else on your network (because it will also be active when connected to your local router)

alias {
  interface "eth0";
  fixed-address 192.168.0.138;
  option subnet-mask 255.255.255.255;
}

I also set a faster timeout and retry, but that shouldn't be necessary.

timeout 5;
retry 60;

And now, that will NOT work... because there is a service that messes it up:
https://linux.die.net/man/8/avahi-autoipd
by overriding DHCP when it fails and assigning it's own "adhoc" ip address in the 169.254.1. net. Mine always came up on .141 but I don't know if that is consistent (couldn't find a way to control it) then you have to reconfigure your PC and even then I couldn't get it to actually communicate on that address. 

So you want to disable the avahi daemon for this to work... but you can't. I tried everything to "neatly" turn it off and it would come back on no matter what. So... I went a bit nuclear on it:

`apt-get remove avahi-daemon `

now, that also removes: 
````
Removing avahi-utils ...
Removing telepathy-salut ...
Removing libnss-mdns ...
````

I don't think we will be doing anything with telephony services (which is what telepathy-salut provides) and the libnss-mdns service is the one that lets you find the device on the .local network. *shrug* I've never used that. 

Now when you start the bot, it takes a minute or two, and then then fixed address will come up and start working. Yeah!

JamesNewton, 2018-09-28:
To gain access to Dexters command line in order to make these or other changes to the Network configuration, a USB connection can be made:
https://github.com/HaddingtonDynamics/Dexter/wiki/Dexter-USB-Connection

cfry, 2018-09-28:
James N,
This is very helpful.
I've just edited that page to make a few things less ambiguous (to me).
Please check for accuracy.

General comment: When giving a list of steps,
clearly separate the steps from one another.
You'll see I did that under the "Putty" section,
preceding each step with a newline and dash
to make it a bullet point.

One thing I don't do enough of when giving instructions is
clearly state what you should expect to see to verify
that a step has been completed successfully.
The goal is to make a user CONFIDENT at each step
that things are going well.

Given that we've got a complex system and have to
document things that may change without our control,
there's no real hope that we can be perfect for everyone
and so we should not struggle too hard to achieve it.
None the less, a bit of redundancy can help our
instructions be *somewhat* resistant to bit-rot.

Since the kind of thing you documented here is related
to the show-stopper connectivity problems I was having
at NYIT hackaton last week, you've done a good job
to help out in this important use case.

JamesNewton, 2018-10-08:
In the end, it's probably better to set the default / starting sd card image up with a simple fixed IP address. If users who are on a large network need to reach multiple Dexters, their networking people can reconfigure them to use DHCP and answer any network security issues at the same time. The fix ip has the following advantages:
- Works out of the box by directly connecting to a local computer via CAT 5 and a known IP. No need to figure out the COM number, install drivers, find the IP address, etc... 
- Avoids issues of network vulnerability by working only with the locally connected PC. 
- Works even when the local network is WiFi only (no CAT 5 connections to the local router).

TODO: Create new image with Fixed IP address.

AndrewSmart, 2018-12-03:
Perhaps [UPnP](https://en.wikipedia.org/wiki/Universal_Plug_and_Play) would interest you? Search for a javascript UPnP library. UPnP can also forward ports on routers which support it (how video games automatically set up connections directly between peers without the user having to configure their router), but it's more of a residential technology. Might help with #31 anyway.

JamesNewton, 2019-02-01:
This is effectively resolved with the new OS image as per #25

closed 2019-02-01 by JamesNewton