HaddingtonDynamics/Dexter #39: Encoder disks difficult to print on some 3D printers

JamesNewton, 2018-10-02:
Because the spacing / gap of the slots on the encoder disks was specifically optimized for certain 3D printers, they can be difficult to print well even on higher resolution printers. If the resolution of the printer / size of the extruded filament doesn't exactly match the design, the slicing can mess up by attempting to get more than one filament pass between some slots and only one between others. 

It may be better to design a disk for laser cutting. 

Or it may be that very high resolution (e.g. resin) printers are a good choice.

CaliTarheel, 2018-10-03:
I can experiment with both.

CaliTarheel, 2018-10-03:
If I can have a DXF file of what the laser cut needs to be, I can experiment with that on some 2.9mm black acrylic. 
[Disk.zip](https://github.com/HaddingtonDynamics/Dexter/files/2444312/Disk.zip)

I have a form 2 and am game to try printing out the attached file.  The file can be read with Preform, downloadable from https://formlabs.com/tools/preform/ so you can see how the supports attach.

JamesNewton, 2018-10-04:
Thanks @CaliTarheel we appreciate it. 

We don't have expertise available on resin printing, so our opinions on where supports should be wouldn't really matter. Probably the best thing to do is just try the print and send it to us for testing. 

I will look into how to convert the files into DXF format for laser cutting. Getting the correct thickness of material may be an issue. The disks need to be about 0.04" / 1mm thick to pass through the optical slot, while still being completely opaque.

JamesNewton, 2018-10-04:
Just to clarify again, the DIFF disks are less interesting because they are smaller and have a lower slot count and mounting them is difficult. The disks that are worth trying are the pivot, base, and end arm disks.

CaliTarheel, 2018-10-04:
James, 

For the supports, the idea is to keep them close to but not touching critical tolerances.  Some sanding or pocking can occur. which is why I've tried to keep them away from the slots and the inner side of the ring.

I'll take out the DIFF disk and see how the other one goes.

If the STL files are accurate, which they'd need to be, I can reverse engineer from those.  Though it looks like it'll be about 6 weeks before I can get the 1mm acrylic needed to test it.

JamesNewton, 2018-10-04:
@CaliTarheel great, thanks!