HaddingtonDynamics/Dexter #50: Documentation for Eye Calibration

JamesWigglesworth, 2019-01-20:
Add doc for when Eye Calibration is needed and how do do it.

JamesNewton, 2019-01-20:
It's documented in DDE. Referenced here:
https://github.com/HaddingtonDynamics/Dexter/wiki/Encoder-Calibration

Is there any other way to do it? I mean, outside DDE? Or is there a reason to duplicate the documentation into the WiKi? Or is the documentation in DDE not sufficient? 

Actually.... now you mention it, knowing when that calibration is needed is a tricky point. How DO you know? What are the symptoms that appear that let you know a joint needs to be calibrated? I guess "freakout" is a good one. Perhaps we need a "troubleshooting" list, with symptoms and solutions?

cfry, 2019-01-21:
James N,
good point about "how do you know when you need calibration?
Is there some quick automated test we could do
that would tell a user
"um, looks like you need to calibrate your robot by doing ...."
It might even run automatically the first time
a user runs a job on dexter during a dde session.
Like printing out a general "health" report!

JamesNewton, 2019-08-28:
The monitor added in recent releases should resolve the "you need to calibrate" issue. When it sees extreme velocities, it will throw an error and DDE can see that and warn the user. @cfry we need to resolve the error handling stuff in DDE to make this work.

closed 2020-11-05 by JamesNewton