HaddingtonDynamics/Dexter #53: 'Left' configs no longer work

JamesWigglesworth, 2019-01-26:
Joint 5 appears to be off by 180° in left configs.
Other configs may not be working either. 
Mostly likely broken in both DDE and DexRun inverse kinematics.

cfry, 2019-01-27:
IS this something that happened in the latest DDE release,
or has it been there for a while?

JamesWigglesworth, 2019-01-27:
I think it's been in there for a while.My guess is ever since we had the confusion over which rotation direction was positive.

Sent via the Samsung Galaxy Note® 4, an AT&T 4G LTE smartphone

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/HaddingtonDynamics/Dexter/issues/53#issuecomment-457940475",

"url": "https://github.com/HaddingtonDynamics/Dexter/issues/53#issuecomment-457940475",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

JamesWigglesworth, 2019-01-28:
Bug fixed.
Will be in DDE release 3.0.8 and later.

To test here's position and direction that work with all combinations of configs:
[0, 0.3, 0.4], [0, 0, -1]

closed 2019-01-28 by JamesWigglesworth