HaddingtonDynamics/Dexter #58: Windows 8 / 10 can't browse SAMBA share

JamesNewton, 2019-02-18:
We hoped that the new Xilinux / Lubuntu 16.04 image would resolve this issue, but it appears to still be a problem.

JamesNewton, 2019-02-18:
This might apply, but I think the version of SMB used in 16.04 is beyond 1.0, and the share still fails to load even if you directly point to it. 
http://wdc.custhelp.com/app/answers/detail/a_id/20736/~/how-to-enable-smb-1.0%2Fcifs-file-sharing-support-on-windows-10

JamesNewton, 2019-02-18:
It may be that it still works but you have to do it via "Add a network location".
https://www.techrepublic.com/article/how-to-connect-to-linux-samba-shares-from-windows-10/

JamesNewton, 2019-02-18:
Enabling SMB 1.0 client support in Windows may be the answer. That and a few other possible issues are listed here:
https://social.technet.microsoft.com/Forums/en-US/e43bdd35-6dc6-4182-b7a4-1b77cb7fc16a/following-upgrade-to-windows-10-nas-unit-does-not-show-under-quotnetworkquot?forum=win10itpronetworking

JamesNewton, 2020-11-05:
Kamino cloned this issue to [HaddingtonDynamics/OCADO](https://github.com/HaddingtonDynamics/OCADO/issues/58)

RonChauffe, 2020-11-05:
Please fix and stop this repeated email issue.