HaddingtonDynamics/Dexter #59: Track and return settings of FPGA registers

JamesNewton, 2019-02-19:
There is currently no way for a control program (like DDE) to know how each of the registers in the FGPA has been set. Many (most) of them are write only, and so DexRun firmware can not query those registers and return their current setting. Also, the default setting of the registers might change, and therefor it can be impossible to know how a register is set unless the firmware or control program has set them. 

Even if they have been set, or if the default settings are moved into the firmware, we do not currently "shadow" or track the last known setting of that register. Adding this functionality will require serious work. 

Finally, some means of returning those settings is necessary. Probably a "fake" JSON file returned via read_from_robot via a "#" name like we did for #XYZ would work and be easy to parse by the controller.

JamesNewton, 2019-08-21:
@JamesWigglesworth we really need to know which registers are important for this functionality, to give us a starting point.

JamesNewton, 2019-10-14:
Added in 
https://github.com/HaddingtonDynamics/Dexter/commit/31c1e41f59eb86452bd60402ce426722c248e1ff

closed 2019-10-14 by JamesNewton

JamesNewton, 2020-11-05:
Kamino cloned this issue to [HaddingtonDynamics/OCADO](https://github.com/HaddingtonDynamics/OCADO/issues/59)

RonChauffe, 2020-11-05:
Please fix and stop this repeated email issue.