HaddingtonDynamics/Dexter #60: job engine to run DDE jobs on Dexter

JamesNewton, 2019-03-28:
Starting this issue to record development history so we know why something was done in the future. The current status (only available on the very latest images) is here:<br>
https://github.com/HaddingtonDynamics/Dexter/wiki/DDE/_edit#job-engine-on-dexter

Goal: Make it easy for DDE on a PC to write jobs to Dexter which are then run ON Dexter. These are one time jobs, with DDE in control of dispatch, so you don't have to SSH in, or depend on SAMBA or anything else. The functionality can even be integrated into DDE. 

And this allows us to start the job ourselves, it doesn't have to be started automatically when Dexter fires up. Experts can figure out how to add an [/etc/rc.local](https://www.raspberrypi.org/documentation/linux/usage/rc-local.md)

First, we must have DDE on Dexter. In this case, we don't need to make a distributable package, we just want to run the source. And having the source directly run means we can develop on Dexter and also use parts of it in other ways (see Job Engine below). So instead of installing the Electron package, we just install the source and use npm to pull in the dependencies. Note: This requires the 16.04 version of the operating system on Dexter.
````
$ git clone https://github.com/cfry/dde
$ cd dde
$ npm i
$ npm run start
````
Of course, the GUI part of the app will only be visible with an X-Server running and since Dexter does not have a video adapter, this must be a remote the [X-Windows Desktop](Dexter-Networking#x-windows). On the current images, an icon is provided to launch DDE from the desktop when logged in via X-Windows. The program takes a while to load (need faster SD Card and interface?) but operation isn't horribly slow.

A "dde_apps" folder for the GUI run of DDE is created under the "/root" folder (alongside Documents, not in it) for the DDE application. Setting the dexter0 ip address to `localhost` in the `/root/dde_apps/dde_init.js` file allows local connection of DDE to DexRun. 

To run DDE jobs without the full DDE GUI interface, e.g. via [SSH](Dexter-Networking#shell-access-via-ssh), you can start them from `~/Documents/dde` with the command:<br>
`node  core  define_and_start_job  job_file.dde`

There are a few things to tweek before that will work:

From /root/Documents/dde, `nano core/serial.js` and comment out the line: `const SerialPort = require('serialport')`. There is some mismatch between the component that manages serial ports on our OS vs others. 

**Job Engine Initialization:** When run for the first time, the job engine creates a `dde_init.js` file in the `/root/Documents/dde_apps` folder. (note this is different than for the GUI DDE on Dexter which is in `/root/dde_init.js`). The job engine defaults to simulate, so the jobs don't actually make the robot move until the dde_init file is edited to add `,simulate: false` after the IP address in the definition of dexter0. The IP address is set to `localhost` so it will work no matter what IP address Dexter is actually assigned.

`/root/Documents/dde_apps/dde_init.js`:
````Javascript
persistent_set("ROS_URL", "localhost:9090") //required property, but you can edit the value.
persistent_set("default_dexter_ip_address", "localhost") //required property but you can edit the value.
persistent_set("default_dexter_port", "50000") //required property, but you can edit the value.
new Dexter({name: "dexter0", simulate: false}) //dexter0 must be defined.
````

Keep in mind the version of DDE on Dexter may need to be updated.  It's 3.0.7 on the initial release of the 16.04 image.

You may want to use
`node  core  define_and_start_job /srv/samba/share/job_file.dde`

When you 'run a job' as defined above, it sets window.platform to "node". If you are in dde, `window.platform == "dde"` will evaluate to true. That means you can customize any code written based on this "platform" i.e.
````Javascript
if(window.platform == "node")      { /* hey I'm in node. */ }
else if (window.platform == "dde") { /* we're in dde! */ }
````

The system software takes advantage of this. One important case is that the "out" function is defined as:
````Javascript
function out(val="", color="black", temp=false, code=null){
    if(window.platform == "node") { console.log(val) }
    else { /* do formatting and print to DDE's Output pane */ }
}
````
Thus when running on node, 'out' only pays attention to its first arg, and it sends the first arg directly to the console.

On the development image, (for the next release) there is an updated /etc/systemd/system path and service which looks for any changes in the /srv/samba/share/job folder and when seen, executes the following script "RunJob" (also in that folder):
````
#!/usr/bin/env bash

cd /root/Documents/dde
for i in /srv/samba/share/job/*.dde; do
        [ -f "$i" ] || break
        echo "running $i"
        sudo node core define_and_start_job $i >> $i.$(date +%Y%m%d_%H%M%S).log
        # must use sudo or node doesnt know who the user is and cant find the dde_apps folder.
        rm $i
done
````
which then fires off the job engine, does the job, and puts the resulting output in a <filename>YYMMDD_HHMMSS.log file.

[... 21 lines cut ...]
cmd_name: define_and_start_job args: /srv/samba/share/job/try.dde
load_files called with: /srv/samba/share/job/try.dde
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;loading file: /srv/samba/share/job/try.dde
Done loading file: /srv/samba/share/job/try.dde
Job: my_job pc: 0 <progress style='width:100px;' value='0' max='2'></progress> of 2. Last instruction sent: {instanceof: move_all_joints 30,45,60,90,120}&nbsp;&nbsp;<button onclick='inspect_out(Job.my_job)'>Inspect</button>
Creating Socket for ip_address: localhost port: 50000 robot_name: dexter0
Now attempting to connect to Dexter: dexter0 at ip_address: localhost port: 50000 ...
Succeeded connection to Dexter: dexter0 at ip_address: localhost port: 50000
Job: my_job pc: 0 <progress style='width:100px;' value='0' max='2'></progress> of 2. Last instruction sent: {instanceof: move_all_joints 30,45,60,90,120}&nbsp;&nbsp;<button onclick='inspect_out(Job.my_job)'>Inspect</button>
Job: my_job pc: 1 <progress style='width:100px;' value='1' max='2'></progress> of 2. Last instruction sent: {instanceof: move_all_joints 0,0,0,0,0}&nbsp;&nbsp;<button onclick='inspect_out(Job.my_job)'>Inspect</button>
Job: my_job pc: 2 <progress style='width:100px;' value='2' max='3'></progress> of 3. Last instruction sent: Dexter.get_robot_status &nbsp;&nbsp;<button onclick='inspect_out(Job.my_job)'>Inspect</button>
top of Job.stop_for_reason with reason: Finished all do_list items.
Job: my_job pc: 3 <progress style='width:100px;' value='3' max='3'></progress> of 3. Done.&nbsp;&nbsp;<button onclick='inspect_out(Job.my_job)'>Inspect</button>
Done with job: my_job
````
I'm happy about the ability to get a job onto Dexter and run it /on Dexter/ using nothing more than DDE. I like that the job file is deleted after it's run. I like that systemd is /apparently/ smart enough not to start the script until the file is done being written (apparently?) and I like that the output is retained. 

What I don't like is that it's retained forever. If people use that alot, it's going to fill up the folder with junk. I see a few easy solutions and some harder ones.
1. When a new job is run, log files from the prior jobs are erased. e.g. add rm /srv/samba/share/job/*.log to the RunJob script. 
2. When DDE "picks up" the log file, it could write back a zero length string to that file name, and DexRun could be changed to make it just delete the file when it is written a zero length string. (I really don't like that idea, it has MANY problems. 
3. Implement shelling out to bash when read_from_robot is called with a `<filename>` string. This on the TODO list anyway, sometime after slaying the dragon. (note: it's on READ not write, because the output of the bash shell gets buffered back to DDE, you write the file, then read the result, which actually triggers running the file). I don't want to do that yet, because it's complex.

Which reminds me, there is currently no way to know what the log file name will be, so DDE can't really read back the result. Since the job file gets whacked anyway, I think the log file should NOT include the date and time and just be <file>.log. So the test.dde job would result in a test.dde.log file, DDE can read that, and (as per option 1) next time you write test.dde, the old test.dde.log file gets whacked automatically and replaced with a new one. That seems like a simple solution, no? 

Last problem, DDE doesn't know when the job is finished. So the log file keeps expanding, and shouldn't be read_from_robot'ed until it's done. I think the solution to that is to always write the log into the file with date and time, then when the job has finished, rename that file to the <file>.log name. DDE can poll and when it returns more than a null string, it can be sure it's getting the entire string.

JamesNewton, 2019-03-29:
Ok, a few changes: 

1. There is a sub folder under job called `run`, and that run folder is what is watched and triggers the service, not the job folder. That allows things in job to be edited / changed without re-triggering the service. 

2. There is a sub folder under job called `logs` where log files end up. 

3. The systemd files were changed as follows:<br>
ddejob.path
````
[Unit]
Description=DDEJob
[Path]
PathModified=/srv/samba/share/job/run
[Install]
WantedBy=multi-user.target
````

ddejob.service
````
[Unit]
Description=DDEJob
[Service]
ExecStart=/srv/samba/share/job/RunJobs
````
and then started with
````
sudo systemctl enable ddejob.path
sudo systemctl start ddejob.path
````
Note: if you edit those, do a 
````
sudo systemctl daemon-reload
````

The RunJob script is:
````
#!/usr/bin/env bash

# first remove prior job logs and junk

cd /root/Documents/dde
for i in /srv/samba/share/job/run/*.dde; do
        [ -f "$i" ] || break
        jobname=$(basename -- "$i")
        echo "running $jobname"
        echo "Logfile for $jobname on $(date +%Y%m%d_%H%M%S)">/srv/samba/share/job/logs/$jobname.tmp
        sudo node core define_and_start_job $i >> /srv/samba/share/job/logs/$jobname.tmp
        # must use sudo or node doesnt know who the user is and cant find the dde_apps folder.
        rm /srv/samba/share/job/logs/*.log
        mv /srv/samba/share/job/logs/$jobname.tmp /srv/samba/share/job/logs/$jobname.log
        # copy large log files only after they are fully written so dde doesnt seem them until complete
        rm $i
        # remove the job file
done
````
so now the DDE job:
````
new Job({name: "test_job_engine", do_list: [
	Dexter.write_to_robot(
    	"new Job({name: \"my_job\", do_list: [Dexter.move_all_joints([30, 45, 60, 90, 120]), Dexter.move_all_joints([0, 0, 0, 0, 0])]})"
        , "job/run/try.dde"
        )
    ]})
```
has the same result, but we can add code to `read_from_robot("my_job_log", "job/logs/try.dde.log")` and when that returns nothing, just loop until it returns the result. 

Seems to work quite nicely.

cfry, 2019-03-29:
Rather than write the file to

"job/run/try.dde"

How about writing it to

"job/run/my_job.dde"

If the script is just looking for any files

in the job/run  folder,'

then it will be able to find it

AND in case something goes wrong,

we can at least easily identify

what job it is.

Same thing for

job/logs/try.dde.log

IE DDE will know the name of the job

it wants to check, so can ask explicitly

for job/logs/my_job.dde.log

But if something is left hanging around, etc.

then we'll be able to debug easier.

_____

I've been doing a bunch of design work on the

DDE side to support this today.

I'd like to be able to send a cmd

to Dexter to stop such a job.

Something like:

Dexter.write_to_robot("stop_job my_job")

To implement, I guess you have to

save away the process id of the node

process associated to the job name.

Then if you receive

"stop_job my_job"

do

kill -9 my_job_process_id

because, as we all know,

a -8 kill sometimes just isn't strong enough :-)

JamesNewton, 2019-08-21:
This has been included on the prior image and on the upcoming one. 

The ability to run BASH shells is a much better / faster way to trigger job engine code if needed, as enabled by #20 (be sure to keep polling via "r 1 `" until the job engine completes") 

Also, if DDE gains the ability to SSH into Dexter, that may provide an even better interface. 

In any case, it's there, it works, it's based on DDE 3.0.7 but that will be updated in the future.

JamesNewton, 2020-03-16:
Changes made in DDE 3.5.6 cause this to not work on Dexter because of a package for 

DDE 3.5.2 works but you must do an `npm run rebuild`

`npm install serialport` may be required before the rebuild? Further testing required. 

For 3.5.2 I did:
````
cd /root/Documents
git clone https://github.com/cfry/dde
cd dde
git checkout v3.5.2
npm install
npm run rebuild
npm run start
````

JamesNewton, 2020-03-28:
Support for serial ports has been difficult with the dde "job engine" on Dexter. For some strange reason, we can't seem to rebuild the serial module for the version of node in use. When we try to `require("serialport")` we get the following error messages:
````
/root/Documents/dde/node_modules/bindings/bindings.js:91
        throw e
        ^

Error: The module '/root/Documents/dde/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 67. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:750:18)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:560:12)
    at Function.Module._load (internal/modules/cjs/loader.js:552:3)
    at Module.require (internal/modules/cjs/loader.js:657:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at bindings (/root/Documents/dde307/node_modules/bindings/bindings.js:84:48)
    at Object.<anonymous> (/root/Documents/dde307/node_modules/@serialport/bindings/lib/linux.js:1:98)
    at Module._compile (internal/modules/cjs/loader.js:721:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:732:10)
````

`npm rebuild` does not help. 

`npm install serialport` does not help.

However, if we start in a fresh folder, and `npm install serialport` we can write little serial programs that work just great, in that folder. Something in the dde file setup is causing the serialport module to stay stuck in the wrong version. 

A complete HACK to solve this is the following:
````
cd /root/Document
mkdir temp
cd temp
npm install serialport
cd /root/Document/dde/node_modules
mv @serialport @save
mv serialport save
cp -r /root/Documents/temp/node_modules/@serialport .
cp -r /root/Documents/temp/node_modules/serialport .
````
having done that, serial ports now work in the Job Engine.

JamesNewton, 2020-08-06:
### WebSocket Job Engine Interface (incl non-browser)

To use the websocket interface outside of the browser environment, potentially from any websocket capable environment and not only the browser, via the user_data variables. This requires a few lines being changed in the /srv/samba/share/www/httpd.js file. Specifically

- Add:
`      serve_job_button_click(browser_socket, mess_obj)`
after the line:
`      console.log("\n\nwss server received invalid message kind: " + mess_obj.kind)`

- Change the else clause at the end of serve_job_button_click to
```
    	let code 
        if(job_name === "keep_alive") { //happens when transition from keep_alive box checked to unchecked
        	code = "set_keep_alive_value(" + mess_obj.keep_alive_value + ")"
        }
        else {
          //code = "Job." + job_name + ".server_job_button_click()"
          //e.g. `web_socket.send(JSON.stringify({"job_name_with_extension": "dexter_message_interface.dde", "ws_message": "goodbye" }))`
            if (mess_obj.ws_message ) { // {"job_name_with_extension": jobname.dde, "ws_message": data}
              code = 'Job.'+job_name+'.user_data.ws_message = "' + mess_obj.ws_message  + '"'
              }
            if (mess_obj.code) {
              code = mess_obj.code
            }
            else {
              code = 'Job.maybe_define_and_server_job_button_click("' + jobfile + '")'
              }
        }
        console.log("serve_job_button_click writing to job: " + job_name + " stdin: " + code)
        //https://stackoverflow.com/questions/13230370/nodejs-child-process-write-to-stdin-from-an-already-initialised-process
        job_process.stdin.setEncoding('utf-8');
        job_process.stdin.write(code + "\n")
        //job_process.stdin.end(); 
````

This allows us to send any "kind" of message (or one with no kind) to the job engine. And if we include a ws_message item in the object we send, it will be inserted into the jobs user_data.

For example:
````js
// File: /srv/samba/share/dde_apps/dexter_message_interface.dde
new Job({
    name: "dexter_message_interface",
    when_stopped: "wait",
    inter_do_item_dur: 2,
    show_instructions: false,
    user_data: {ws_message: "hello"},
    do_list: [
        Control.loop(true, function(){
        	if (this.user_data.ws_message) {
              out(this.user_data.ws_message)
              this.user_data.ws_message = undefined
              }
        	})
        ]})
        
````
Note: 
- The jobs file name must be the same as the job name. 
- The `show_instructions: false,` setting will keep the communications channel from being swamped as instructions execute.
- Setting `inter_do_item_dur: 2,` will slow down the job so you can see it operate. You should probably set that to a smaller number for better performance.

To use this, a web socket connection must be opened to Dexter on port 3001. e.g. 
`ws://192.168.1.142:3001/`

A large number of status and informational strings will be returned from the job engine to the onboard node server. To pick out the ones that were sent back from your job via "out", look for data wrapped in "<for_server>" tags, with a JSON object and a "kind" of "out_call". The data will be in the "val" attribute. e.g.
` <for_server>{"kind":"out_call","val":"APP:hello.","color":"black","temp":false,"code":null}</for_server>`

To send data from the program that opened the connection into the job engine job, send a JSON formatted string through the web socket connection, with the name of the job file, and a "ws_message" string. For example: 
`'{"job_name_with_extension": "dexter_message_interface.dde", "ws_message": "goodbye" }'`
The job should echo that back to you. 

In some environments, the web socket connection will time out and be closed automatically. To avoid that, you can send a JSON string with a "kind" and job name of "keep_alive". Like this: 
`{kind: "keep_alive_click", job_name_with_extension: "keep_alive", keep_alive_value: is_checked}`
This will be processed and won't actually do anything.