HaddingtonDynamics/Dexter #62: AdcCenters not updated in FPGA when written to file.

JamesNewton, 2019-04-17:
When a human is calibrating the joint eyes on Dexter via DDE, they might expect that when the new centers are saved to the robot, they have effect immediately. This is NOT the case. The robot must be power cycled for them to be read into the FPGA from the file DDE writes to the robot.

Commit https://github.com/HaddingtonDynamics/Dexter/commit/0d766de6320b2e64755c999c5daa912108d5f2c9 resolves this by detecting an updated AdcCenters.txt file IF written via write_to_robot and then re-loads that file immediately. 

DDE should still advise the use to restart the robot because there is no way to know if this new function has been implemented in Dexter.

cfry, 2019-04-18:
> Assigned #62 <https://github.com/HaddingtonDynamics/Dexter/issues/62> to
> @cfry <https://github.com/cfry>.

How about if each Dexter has a Manufacture date on it.
We'll know what software was shipped on its SD card,
then we can know what to tell the users.
Maybe manufacture date is something DDE could retrieve
from a file of "specs" on dexter.

JamesNewton, 2019-04-18:
The issue is that read from robot isnt supported on early Dexters and will
actually crash some of them.