HaddingtonDynamics/Dexter #65: Pictures

JamesNewton, 2019-05-14:
Just a fake issue to make it easy to add pictures to the Wiki.

JamesNewton, 2019-05-14:
[image]
[image]

closed 2019-05-14 by JamesNewton

JamesNewton, 2019-05-15:
[image]

Kenny2github, 2019-05-16:
Transparent
[image]

JamesNewton, 2019-05-28:
[image]

JamesNewton, 2019-06-04:
Picture of clamping Dexter base to a table

[image]

JamesNewton, 2019-06-04:
Screwing a Dexter to the table
[image]

JamesNewton, 2019-06-04:
Hot gluing Dexter to a table
[image]

cfry, 2019-06-04:
Crystal clear picture of screwing!
For the hot glue, how about a picture with
one hole filled with glue and the glue gun poised to
squirt in glue to the 2nd (as yet unfilled).
It would be really good if the color of the glue
was not the same color as a foot.

Most pictures in most "assembly manuals"
are so low res as to be pretty ambiguous as
to what you really need to do.
This aren't!

JamesNewton, 2019-06-05:
Hotglue in a foot
[image]

cfry, 2019-06-05:
In the screwing picture, there are only 2 holes in a foot.
Here it looks like there's 4, but in any case
this new black and white picture is confusing,
much more so than the color screwing picture.

jonfurniss, 2019-06-05:
There are two holes for screwing into on the top of the robot foot.
There are four indents underneath the foot that can be filled with glue to
use for additional traction.
While the two holes for screwing line up with the front two glue indents,
the two features are for different purposes and not related.

cfry, 2019-06-05:
Thanks. Now the challenge is to take a picture that
let's you know you're even looking at the bottom of a foot.
Also, I am guessing that you have to
dab glue onto 2 or 3 feet at once then
plunk the whole robot firmly on the table.

JamesNewton, 2019-06-13:
Image of shipping case with first layer / base
[image]

JamesNewton, 2019-06-13:
Where to grab Dexter out of the packing case.
[image]

JamesNewton, 2019-06-13:
Correct jumper settings on MicroZed board
[image]

JamesNewton, 2019-06-18:
Bad encoder eye. LED pulled out of encoder block.
[image]

JamesNewton, 2019-06-18:
Good encoder, after replacing LED in hole of encoder block.
[image]

JamesNewton, 2019-06-18:
LED or Photo sensor pulled out of hole in encoder block
[image]

cfry, 2019-06-18:
It seems to me that the main hardware problems with Dexter are
with the encoders and their optical sensors, is that right?

JamesNewton, 2019-10-25:
This is what you should see on the console after installing a WiFi adapter. Note that it does not have an IP address, but it is shown in the list of adapters, after eth0 and lo. 
[image]

JamesNewton, 2019-10-25:
Opening screen of `nmtui`
[image]

JamesNewton, 2019-10-25:
Select Activate a connection in `nmtui`
[image]

JamesNewton, 2019-10-25:
Selecting an SSID in `nmtui`
[image]

JamesNewton, 2019-10-25:
`ifconfig` now shows an IP address for the WiFi adapter.
[image]

JamesNewton, 2019-11-20:
Correct orientation for insertion / removal of tools from the Version 2 Tool Changer. 
[image]

JamesNewton, 2020-03-20:
Re-inserting the SD Card. SEE:
https://github.com/HaddingtonDynamics/Dexter/wiki/SD-Card-Image
for critical instructions!

[image]

JamesNewton, 2020-06-03:
Dexter HDI packedin case:
[image]

JamesNewton, 2020-06-05:
Dexter HDI, removal from packing case
[image]

JamesNewton, 2020-06-05:
Alignment mark on J1 on Dexter HDI
[image]

JamesNewton, 2020-06-05:
Dexter HDI with skin showing expected blue light.
[image]

JamesNewton, 2020-07-28:
Dexter FPGA control system block diagram.
[image]

JamesNewton, 2020-07-28:
Dexter Control System, "Keep Position" section expanded. Note that integration is provided by the stepper motor /after/ the PID section. 
[image]

JamesWigglesworth, 2020-08-26:
[image]

JamesNewton, 2020-09-16:
Animation of one encoder sensor as the slot passes the mask.
[image]

JamesNewton, 2020-09-16:
Animation of the encoder showing the phase between the two sensors. 
[image]

JamesNewton, 2020-09-23:
Diagram of link lengths 4 and 5

[image]

JamesNewton, 2020-10-15:
[image]

JamesNewton, 2020-10-15:
[image]

JamesNewton, 2020-10-15:
Dexter HDI Joints Range Of Motion
[image]

JamesNewton, 2021-03-11:
[image]

JamesNewton, 2021-10-01:
[image]