HaddingtonDynamics/Dexter #79: SD Card Image

topteamc, 2020-02-11:
hi,
first i want to thanks every one participate in this awesome project.
can get the SD card image?
thanks again.

jonfurniss, 2020-02-11:
Hello, here is a link to the most recent version of the Dexter HD image:
https://drive.google.com/open?id=1cT9Z0NyPovMRyLEJriLUFKsDkP9Zk0cy
The zipped file is about 2gb, so might take a while to download!
If you have any questions that need a quick response, feel free to join our Discord channel: https://discord.gg/ZWpeVve

JamesNewton, 2020-02-11:
Just FYI, more information on the SD card image is available here:
https://github.com/HaddingtonDynamics/Dexter/wiki/SD-Card-Image
and documentation on the setup of the image is included in #25

closed 2020-02-11 by JamesNewton