HaddingtonDynamics/Dexter #81: Not able to interface the dexter arm with my PC having ubuntu 18.04

sumr4693, 2020-05-16:
Hi,

I have not worked related to ethernet devices before, and so I have difficulties in interfacing my PC with the Dexter robot arm. I followed the instructions given in the website:

1. I set my ethernet adapter IP address to : 192.168.1.100 (for example) and subnet mask: 255.255.255.0
2. I connected the CAT 6a (tried also with new CAT 5e cable as well)  ethernet cable, which I found in the robot arm kit, between PC and dexter arm.
3. I tried to ping the dexter arm (whose default address is supposed to be 192.168.1.142) from both the command terminal and DDE interface. But I didn't get any response.
4. I tried the above steps for IP address for '0-255 in the last section for both 192.168.1.* and 192.168.0.* as well. But no response again.
5. I also tried by reinserting the mounted SD card in the board, and by switching off my WiFi when pinging the arm. But nothing worked.

Is there someway I can find the dexter ip address either from the physical connection, or by using and reading from the SD card? Please help me to fix this.

Thanks in advance!

JamesNewton, 2020-05-16:
Please try the USB connection as per:
https://github.com/HaddingtonDynamics/Dexter/wiki/Dexter-USB-Connection
that will work IF the redboard is booting. Also check the troubleshooting page on the wiki to verify you are getting the expected lights on the board when it boots.

sumr4693, 2020-05-16:
Thank you very much for your immediate response! Ok I will try these and let you know.

sumr4693, 2020-05-18:
I made the USB connection between dexter and PC, and got the following boot commands from the dexter:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

From this info, will I be able to find anything useful to figure out the ethernet communication issue? Thanks in advance!

sumr4693, 2020-05-18:
Also, I checked the LEDs status when booting:
Green: steady 'ON' (Power LED)
Blue: steady 'ON' (related to FPGA)
Red: 'ON' for few seconds and then 'OFF' (related to disk access to SD card)

Is Red LED functioning a normal behaviour?

sumr4693, 2020-05-18:
Regarding the jumper configuration:

To boot from SD card, I saw that the jumpers should have the connections JP1: 1-2, JP2: 2-3, JP3: 2-3

And this is the image with the jumper connections:
[image]

The markings JP1, JP2 and JP3 in the microzed board seen above, are on the different locations from the board which is seen in Troubleshooting page. And I could see "1" on either sides of the three pins for each jumper, so it's confusing to locate the first pin for each jumper. I hope that the board in the attached image is the revised version.

Kindly clarify how to locate the numbers 1,2,3 for each jumper.

Thank you!

JamesNewton, 2020-05-18:
Now I'm confused. I'm not aware of any changes to the labeling on the microzed board. Can you post a picture of your board showing that area? I'm checking with the group to see if they are aware of a change... what name did you order the kit under?

JamesNewton, 2020-05-18:
In any case, based on your prior post of the USB data, it appears the jumpers are in the correct location, but the SD card isn't working correctly. e.g. it can't mount the ext partition and start the OS. It IS able to read the FAT partition and start the .bit file. I would re-burn the SD card from the image or contact sales for a replacement.

sumr4693, 2020-05-19:
Thanks for figuring out the issue! I connected the SD card via card reader to my PC, and I am seeing a 16 GB volume drive (the list of files: boot.bin, devicetree.dtb, DexRun.c, uImage, xillydemo.bit and a folder - System Volume Information) and a 8 GB volume drive (I think this may be the ext partition)  
1. I don't find the details from the "SD card image" page on what the image file does look like. I will be using Etcher to flash the image, since my PC is Ubuntu based. Does Etcher automatically take care of the ext partition in the SD card? Is it just enough to add the flash file and select "Flash"? Or do I need to delete the contents or format the SD card?
2. In the "SD card image" page, it is mentioned to contact you to get the image file. Can you please provide the image file? 

This is the 16 GB volume drive screenshot:
[image]
This is the 8 GB volume drive screenshot:
[image]

If this reburning doesn't work either, my company supervisor will be contacting sales team of Haddington Dynamics.

Thank you!

sumr4693, 2020-05-19:
This is the microzed board image from the troubleshooting page:
[image]
And this is the microzed board I have in the robot arm:
[image]

If you see both the images, the markings of JP1, JP2, JP3 are on the opposite sides. So I was wondering if there is any chance that the numbers 1,2,3 could also be reversed.

And, for your information, this robot arm was exported on May 9, 2019 from Haddington Dynamics. I am just a master thesis student, so I don't know regarding the order details. But the invoice has "Industrial Robots, NESOI" under description. Does this help?

Thanks!

JamesNewton, 2020-05-19:
Can you email jamesnewton@hdrobotic.com so we can email you a link to the SD card image? And also, in the email please include what institution you are with and who your supervisor is so we can look this up in the sales db and provide appropriate support? Thank you. I will close this issue and we can continue support via email. I'm assuming something was damaged on the SD and it should be replaced rather than just re-imaged, but we can try that first.

closed 2020-05-19 by JamesNewton

sumr4693, 2020-05-20:
Sure! I will contact you. Thank you for providing the email id!