HaddingtonDynamics/Dexter #94: Support Dynamixel XC430 on Tool Interface

JamesNewton, 2020-06-30:
The XL-320s have a limited rotational range which sometimes limits our ability to do things. The XC430 doesn't cost much more and provides higher torque, stronger body, and continuous rotation modes. It is, however, physically larger and so is not a direct replacement. It also has a completely different memory map and more complex operating modes. 

There is a new #Servo # Addr Len for the 'r' (read from robot) oplet that lets you query a servo directly and get back the result. I'm hoping it might be helpful for reading back torques really quickly (faster than the monitor mode) or for diagnosing Servo issues. e.g. 
````
r 0 #Servo 1 32 2; Read Max Voltage (currently returning A0 or 160 or 16 volts)
r 0 #Servo 1 34 2; Read Min Voltage (currently returning 3C or 60 or 6 volts)
r 0 #Servo 1 70 1; Read Hardware error (returns 1 which is "Input Voltage Error")
````
Sending an `S RebootServo 1;` does not resolve the error. Sending an `S ServoSet2X 1 34 59;` Set Max Min voltage 0x22 0x3B gets me a packet status error of 0x84 which is "Data Range Error: Data to be written in the corresponding Address is outside the range of the minimum/maximum value"

So it's looking like our days of running the servos at 5 volts when they are spec'd for a min of 6 volts are over. Supporting the XC430's will apparently require another power supply... Not sure why the XL320's were willing to work around it and the XC430's don't seem to be.

We do, as it happens, have an extra power supply ready to go:
https://workspace.circuitmaker.com/Projects/Details/James-Newton-2/Dexter-Tool-Interface-Servo-Power-Supply
But it's another item on the BOM, it's hard to find a place to mount it, and getting them made is a pain. When I designed it, I looked for anything we could just buy from China, but didn't find anything that met the current requirements. That may have changed by now, so we should certainly look again.

Useful instructions:

````
r 0 #Servo 1 32 2; Read Max Voltage
r 0 #Servo 1 34 2; Read Min Voltage
r 0 #Servo 1 70 1; Read Hardware error

S ServoSet2X 1 32 160; Set Max Max voltage 0x20 0xA0
S ServoSet2X 1 34 59; Set Max Min voltage 0x22 0x3B

S RebootServo 1; Reboot servo 1
````

JamesNewton, 2020-07-30:
Change `S RebootServo _id_` to take `S RebootServo _id_ _type_` e.g. `S RebootServo 2 430` which adds a new servo to the list to be monitored with a type of XC430. The monitor has been updated to monitor up to 20 servos, but it only checks one per monitor loop. The loop speed has been increased to once every ~5000us. So if there are only 2 servos, they get checked a bit faster than they did before (was 15000) but if there are 10, the monitor doesn't lock up, it just checks each one every 500us or so. 

Of course, there is currently NO way to get the data collected for any servo address other than 1 or 3 (joints 7 or 6) back from the robot, but that will come. Probably need to expand `r 0 #Servo` to return all monitored servos positions and loads, and `r 0 #Servo _id_` to return the position and load of that servo. 

Currently, @jonfurniss has shown that the Physical User Interface works for recording and playback with the 430's, but there are issues because the units are wrong. If you tell a 320 to move 1, that is 0.29' but a 430, it's 0.088' The plan is use JointsCal and HomeOffset values to correct for those units. This correction is now in place, but there is no way to set numbers into those correction factors because the `S AxisCal ...` and `S HomeOffset ...` only take 5 parameters. 

Perhaps the RebootServo should optionally accept JointCal and HomeOffset values. e.g. `S RebootServo 2 320 1 0` would add a standard XL320 expecting 320 units but `S RebootServo 3 430 11863 148` would setup a 430 with an offset of 148 ('cause DDE LTS expects that for J6) and a multiplier of 11863/arcseconds = ~3.295 because if you tell a 320 to move 1, that is 0.29' but a 430, it's 0.088' and 0.088*3.295 ~= 0.29' see? Call that "compatibility mode" 

Then if you want better accuracy, you do `S RebootServo 3 430 1 0` and you have 0.088' control, but you have to work around DDE's messing around with the values. 

And should this change EESpan and EERoll? Or just 'a' / 'p' / etc... ? (I think both, right?)

JamesNewton, 2020-08-12:
### Monitoring

Update monitor thread to read which ever servos have been rebooted, according to the type supplied. Supports up to 20 servo id's or rather servo id's 1 to 20. Only ID's 3 (J6) and 1 (J7) are currently readable via the standard status responses. 

Possibilities 
- Add additional g # status_mode's with one that only returns status data for the servos
- As above, but use the currently unused "completed time stamp" field to send in the version of the status you want back, without changing status_mode, so all the other commands will return the standard status (or which ever you selected with g # )
- As above but the special return status is a JSON array, instead of binary data.
- Add some options to the `r 0 #Servo ID, Addr, Len` read from robot command so that if only an ID is specified (no addr or len) then all the available monitor data for that servo is returned. 
- As above but if no ID, addr, or len is supplied, return the ID and current position and load of all monitored servos.

JamesNewton, 2020-09-04:
For joints 6 and 7, the standard move all joints and other commands work. For additional servo id's we need to send ServoSetX commands to set the goal position. James W wrote and tested this code. It translates the desired position setting to the hex format needed to send a binary value. 

````js
function num_to_hex(num, size = 12){
    let num_array = Vector.make_matrix(1, Math.round(2 * size / 3))[0]
    let num_str = num.toString(16)
    for(let i = 0; i < num_str.length; i++){
        num_array[i] = num_str[num_str.length - i - 1]
    }
    let str = ""
    for(let i = 0; i < num_array.length ; i += 2){
        str += "%" + num_array[i + 1] + num_array[i]
    }
    out(str)
    return str
}

function set_430_angle(id, angle){
    let size = 12
    return make_ins("S", "ServoSetX", id, 116, size, num_to_hex(Math.round(4096 * angle / 360), size))
}

new Job({
    name: "test",
    do_list: [
        set_430_angle(4, 0),
        Robot.wait_until(1),
        set_430_angle(4, 1*90),
        Robot.wait_until(1),
        set_430_angle(4, 2*90),
        Robot.wait_until(1),
        set_430_angle(4, 3*90)
    ]
})
````

The Servo at ID 4 was setup in Defaults.make_ins with
````
S, RebootServo, 4, 430, 1296000, 0; Set 430 ID 4 for 430 units
z, 1000000; Give it time to reboot
S, ServoSet, 4, 65, 1; Turn XC430 LED on
S, ServoSet, 4, 64, 1; Turn XC430 torque on
z, 1000000; Give the operator time to see it
S, ServoSet, 4, 65, 0; Turn XC430 LED off
````

JamesNewton, 2020-10-08:
Note for the future: If you run an XC-430 with commands for an XL-320, it will probably cause the servo to start reporting a hardware error, the light will blink, and it will refuse to operate. 

This happens because the goal position address for the XL-320 overlaps the Temperature Limit setting for the XC-430. And you can set a temperature limit of as little as 0'C. The servo will then only operate if it's frozen.

To verify this is the case, read back address 31 with  `r 0 #Servo 1 31 1` aka `read_from_robot("#Servo 1 31 1")` and the value returned after the `... 55 80` is the temp limit. Replace that first number with the address of the servo: 1 for J7 (as shown in the example) or 3 for J6. 

To recover from this error, send `S ServoSetX 1 31 80` (again set the first number to the servo address) and verify by reading back that the value was set. Now you can RebootServo (or just power cycle) and everything should be ok.

jonfurniss, 2020-10-09:
DDE currently has a built in offset when sending move all joints to J6 (servo ID 3) of 148° or 510 XL-320 (148°/0.29°/Dynamixel step) units. So a command of 0° from DDE has the XL-320 in the middle of its movement.

When operating the XC-430s from DDE in compatibility mode, we need to account for the larger motion range of the XC-430 as well as the built in DDE offset. When parsing the RebootServo command, DexRun.c first assigns the slope, then the offset, so we need to do the offset with respect to XL-320 units, **not** XC-430 units. Dexter's 0° position is 180° off from the XC-430's 0 Dynamixel unit position, so 180°/0.29°/XL-320 step ~= 620 Dynamixel units. Then 620-510 = 110 is the offset in Defaults.make_ins that has the J6 (servo ID 3) XC-430 behaving as expected.

For setting up a gripper with Servo 3 as joint 6 and Servo 1 as joint 7, paste the following code at the bottom of Defaults.make_ins.
```
S, RebootServo, 1, 430, 4270909, 0;
z, 1000000;
S, ServoSet2X 1, 64, 1; Turn J7 XC430 torque on
S, ServoSet2X 1, 65, 1; Turn J7 XC430 LED on
S, RebootServo, 3, 430, 4270909, 110;
z, 1000000;
S, ServoSet2X 3, 64, 1; Turn J6 XC430 torque on
S, ServoSet2X 1, 65,0; Turn J7 XC430 LED off
```