HaddingtonDynamics/Dexter #110: Consequetive job execution limit (Dexter HDI)

devWork1, 2021-02-11:
Hello,

I have run into what I am hoping is an oversight on my behalf. 

After supplying power to the dexter and having the init established for the correct ip address, the robot (Dexter HDI) begins what I am assuming to be a startup routine. It takes about two minutes to complete and consists of a slight rotation of the shoulder and arm joints with additional articulation of the wrist joints (is there a preferred starting position for the robot upon startup?).

Upon completion the robot assumes an upright condition, does a wrist wiggle, then is ready to accept evaluated jobs. 

_**I have run into an issue where I can only run a single job twice before the robot does a wrist wiggle and fails to start the job a third time.**_ 

I have been able to increase the number of jobs it can perform by creating two different jobs and alternating between the two. However upon the 2nd completion of any of the jobs, any additional jobs fail to send to the robot (leaving a process bar with '0-of-n' items sent.

Additionally, the GUI will self start a calibration job. Whether the calibration job is self initiated or I attempt to launch this job, the calibration will never complete, rather is doesn't do much other that say it is ongoing. Not sure if this is related or not, but looking for some feedback on this as well.

I had read a mention of a 30 second timeout. Is that what is happening? If not how do I resolve this issue?

I am looking to have the robot remain in an idle state waiting for a notification to perform task for extended periods of time (days).

Additionally, if this can be performed all from a command line or python that would be even better.

Thank you for any feedback.

JamesNewton, 2021-02-11:
HI devWork1 (would love to know your name and company)

The 30 second timeout is unrelated. If a connection is made via the raw socket and no communication is sent for 30 seconds, the connect will close automatically. This is not an issue because it can be re-opened instantly. It just means that if an application connects to Dexter, then dies, that will not prevent Dexter from connecting to some /other/ application once the 30 seconds is up. 

The power on motions sounds correct, but there is no need to initiate a connection; Dexter should do that no matter if a PC is connected or not. It is searching for home position. 

When you say "the GUI will self start a calibration job" can you explain what you mean by that? Does a dialog appear on the screen? Does the robot start going through the power on sequence (finding home) without the GUI showing anything?

Missing information:
- What version of the GUI (DDE) are you using? This is displayed in the output pain when you start DDE.

Also, we can help you far more effectively via our Discord server. You do not need to install the Discord app, you can simply use the web service. The "invite" is on our may web site:
https://www.hdrobotic.com/
under "Contact" on the right side. 

Because your model of Dexter (the HDI) is NOT open source, (the prior HD and Dexter 1 models are) there are things we can not discuss on this channel.

cfry, 2021-02-12:
"What version of the GUI (DDE) are you using? This is displayed in the output pain when you start DDE."
Or more easily the title bar of the DDE window.

Also, send us the job you are attempting to run.

JamesNewton, 2021-03-20:
After troubleshooting, problem was shown to be this:
- Internal job engine was running the "find index eyes" job on startup as expected.
- After the job stopped, a significant delay was introduced by a missing 'localhost.localdomain' entry after 127.0.0.1 in the /etc/hosts file which resulted in node waiting for DNS to timeout before running the next job. During this time, the DexRun firmware was free for use by DDE.
- The next job engine job (PhUI) was then eventually started when the users job in DDE had finished, and it held on to the firmware from then on out. 

Notes to future users: Verify that you are NOT in PhUI by attempting to tip up the end effector two "clicks" after bootup if DDE can't connect. If you don't wish to start PhUI on startup, edit the autoexec.jobs file (or the RunDexRun file if that doesn't exist) to remove PhUI form the list of jobs to run on startup.

closed 2021-03-20 by JamesNewton