HaddingtonDynamics/Dexter #111: Outdated Motor Board GERBS

HamzaSI7, 2021-04-07:
The Gerbs files available on the github for the Motor Control PCB and the BOM do not represent the current version of the board.  A board made out of the available files does not work with the MicroZed and causes it to overheat. Can you upload updated versions of the Gerbs and BOM? 

Thank you.

JamesNewton, 2021-04-07:
They are not out of date. The error must be elsewhere.

closed 2021-04-07 by JamesNewton

JamesNewton, 2021-04-20:
Update: My apologies, I was completely wrong. The files we were linking to in the kgalspark archive were out of date and the correct files had somehow not been uploaded to this repo as I had thought. They have been added now:
https://github.com/HaddingtonDynamics/Dexter/tree/Stable_2020_02_04_ConeDrive/Hardware/Motor%20PCB